LumaTeam/Luma3DS #1566: Automatically copy boot.firm and boot.3dsx to the SD if a card is detected without them

hifinerd, 2021-01-09:
A feature I would like to see for those with multiple SD Cards is the ability to store a backup of boot.firm and boot.3dsx on the nand, or just use the existing boot.firm used when booting from CTRNAND to automatically put boot.firm and boot.3dsx on an SD that does not have them, or the copies on the SD are corrupted.

urherenow, 2021-01-09:
Technically, nothing is stopping you from doing that already with GodMode9 (placing a copy on the nand).

But then you would always need to update the NAND files with Godmode9, and Luma/Rosalina settings would also have to be on NAND for the cards with missing files, in which case have any of those files on any card at all would be pointless.

Don't think it will happen.

RS55, 2021-01-09:
This is an issue for Boot9Strap or Fastboot3DS that of Luma

TuxSH, 2021-01-11:
I'll do something like "automatically copy boot.firm" to NAND when upgrading Luma, in future releases.

hbmenu doesn't really work without SD (isn't intended to -- remember, boot.firm on NAND is only meant to be a failsafe mechanism) so there's that too.

closed 2021-01-11 by TuxSH