LumaTeam/Luma3DS #1567: Luma 3DS won't update

ronaldgameking, 2021-01-10:
**System model:**
Old 3DS XL
[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
Sys 11.14.0-46E
[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
BS9

**Luma3DS version:**
10.0.1
[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

No EmuNAND
Screen brightness: ( 4 )

Splash: ( Off  )

Splash duration: ( 1 )

PIN lock: ( Off )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )
No EmuNAND

Use EmuNAND FIRM if booting with R: ( )
No EmuNAND

Enable loading external FIRMs and modules: ( off )

Enable game patching: ( )

Show NAND or user string in System Settings: ( X )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: (  )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

Luma3DS does not update even when files on SD card are replaced, booting from the file on SD card will boot the updated Luma3DS

**Steps to reproduce:**

1.
Replace the `boot.firm` and `boot.3dsx` on the root of the SD card
2.
Put the SD card back into the console. Boot normally to see a ARM11 exception as it didn't update.

**Dump file:**

ronaldgameking, 2021-01-10:
Crash dump
[crash_dump_00000000.zip](https://github.com/LumaTeam/Luma3DS/files/5793463/crash_dump_00000000.zip)

CatsLover2006, 2021-01-12:
B9S might be booting to the internal NAND, try updating to Fastboot3DS using Luma3DS’s plugin loader (hold start on boot).

Sly2141, 2021-01-14:
@CatsLover2006 I've run into a similar issue as OP. Looking at the Luma plugin loader mine says booting from CTRNAND so I think this is my issue is what you're saying. I was going to try your fix but I don't see an option in the plugin loader that matches what you're talking about. Could you be more specific as to what I would need to do to use Fastboot? I don't want to just go pushing buttons And help a brother out whos dumb af when it comes to this.

ronaldgameking, 2021-01-14:
> 
> 
> B9S might be booting to the internal NAND, try updating to Fastboot3DS using Luma3DS’s plugin loader (hold start on boot).

This never happened normally, tried to copy to CTRNAND. Fastboot should **NOT** be required to update/boot from new version!

Steginator, 2021-01-16:
i have the same problem i have the arm11 error. I accidentally made a software update without updating Luma3ds beforehand. I have already replaced the files on my SD card with v10.2.1 but that didn't help. Please help me !!

Smoreslickgores, 2021-02-05:
I think you might need the both top Luma3ds update thingy here watch this video https://www.youtube.com/watch?v=37LSonTDL9M&ab_channel=sthetix

CatsLover2006, 2021-02-08:
> @CatsLover2006 I've run into a similar issue as OP. Looking at the Luma plugin loader mine says booting from CTRNAND so I think this is my issue is what you're saying. I was going to try your fix [...]

Press the home button on boot up to open the Fastboot3DS menu, and change your boot option from the CTRNAND boot.firm to the SDMC boot.firm

closed 2022-06-05 by suppai