LumaTeam/Luma3DS #1570: NTR CFW crashes

BigDave279, 2021-01-15:
I installed NTR CFW and when i run the app it automatically tries to run 3.2 and then it crashes.

urherenow, 2021-01-15:
This isn't the NTR github page.

closed 2021-01-16 by TuxSH

Christchex, 2021-01-19:
Not to be a bother but I assume you guys have seen this:

https://github.com/Nanquitas/BootNTR/releases/tag/v2.13.4

Dimensional, 2021-01-20:
That developer also worked on getting bootNTR integrated into Luma with his own variant. He tries to keep matched with our Luma here, so the latest release has the same version number and was released only a little while after this one. Not sure how stable it is compared to here, but if you have the option to do so with Fastboot3DS, you can switch between versions when needed. Just be sure to know which version you are using if submitting bug reports.

https://github.com/Nanquitas/Luma3DS/releases