LumaTeam/Luma3DS #1573: ‘Force Wireless Connection’ bug

25zcj01, 2021-01-21:
**System model:** N2DSXL

**SysNAND version:** 11.14.0-46U

**Entrypoint (How/what you're using to boot Luma3DS):** B9S

**Luma3DS version:** Nanquitas’ v10.2.1 build

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (Before Boot)

Splash duration: (1 sec)

PIN lock: (8 digits)

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: (y)

Enable game patching: ( )

Show NAND or user string in System Settings: (y)

Show GBA boot screen in patched AGB_FIRM: (y)

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: (y)

--

**Explanation of the issue:**

After forcing a wireless connection via Rosalina, (connection slot 1), restoring the slot makes the setting to enable/disable the power button display and function as if it were the ‘Force Wireless Connection’ setting, and will continue to do so until a reboot

**Steps to reproduce:**

1. Force a wireless connection to any slot

2. Restore slot

3. Look on in horror as the option to disable to power button is now gone 

**Dump file:** N/A

TuxSH, 2021-01-27:
Thanks, good catch!

closed 2021-01-27 by TuxSH

25zcj01, 2021-01-28:
No problem!

commit 3afecb06 (2021-01-27): rosalina: fix #1573