LumaTeam/Luma3DS #1574: Issue with connection forcing

clauz07, 2021-01-22:
When I try to force a connection with Rosalina it doens't work, I need to switch off, and then switch on the wireless communication, and then it works

Oreo639, 2021-01-22:
> When I try to force a connection with Rosalina it doens't work

Wdym? How does it not work? What happens when you enable it? What are you expecting to happen when you enable it?

clauz07, 2021-01-24:
When i try to force the connection it says ''failed to connect'', then if I turn off and then on the wireless connection it says ''successfully connected''

X7ox, 2021-01-27:
Odd, for me it often takes forever to force one but never fails. May mean that something may be wrong with a connection. I use a Wifi Save Editor that allows me to switch wifi slots on the fly without going to settings and performing a connection test.

closed 2022-06-05 by suppai