LumaTeam/Luma3DS #1578: New 2Ds XL Crashes when trying load Miiverse / Nintendo eShop

NintendoPSproGS, 2021-01-25:
[crash_dump_00000003.zip](https://github.com/LumaTeam/Luma3DS/files/5867471/crash_dump_00000003.zip)

The crash happends when trying to load certain 3ds apps (like the two i said on the title) do you know how to fix it? Because i want to revive Miiverse

X7ox, 2021-01-27:
Did you region change your device?

NintendoPSproGS, 2021-01-27:
No, i didn't.

NintendoPSproGS, 2021-02-01:
Hello? I need help, please!

urherenow, 2021-02-01:
GitHub is not a help forum. This space is for reporting real bugs, or making feature requests.

Considering Nintendo discontinued miiverse nearly 4 years ago, this is NOT a bug.

NintendoPSproGS, 2021-02-02:
> GitHub is not a help forum. This space is for reporting real bugs, or making feature requests.
> 
> Nintendo discontinued miiverse nearly 4 years ago, so I don't even know what you expect there. Open a thread on gbatemp, or a discord channel.

I mean that i was gonna use rverse to revive the miiverse services, but if i can't log in, how am i supposed to do it?

closed 2021-04-03 by NintendoPSproGS