LumaTeam/Luma3DS #1580: Amiibos arent being read when launching with sd card

onapjd, 2021-01-26:
**System model:**
New 3ds XL
[New 3DS XL]

**SysNAND version (+emu/redNAND version if applicable):**
11.4.0-46U
[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
"Booted from SD via B9S
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
v10.2.1 stable
[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: 4(x)

Splash: off(x)

Splash duration: 1(x)

PIN lock: off(x)

New 3DS CPU: off(x)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
Amiibos dont read when booted from sd (acts as if no amiibo is there despite there being one). When booted from CTRNAND
via B9S, amiibos are read just fine (no difference in the luma settings)

**Steps to reproduce:**

1. Boot with sd card and nav to amiibo settings, any option, place amiibo, doesnt read

2. Boot without sd card and nav to amiibo settings, any option, place amiibo, does read

**Dump file:**
No dump

TuxSH, 2021-01-26:
Can't reproduce.

Are you using something like Wumiibo? Try disabling game patching.

onapjd, 2021-01-26:
No im not using wumiibo. And disabling game patching fixed it, however using amiibos with game patching enabled hasn't caused an issue like this

TuxSH, 2021-01-27:
You must have something in /luma/titles then. The issue is on your end.

closed 2021-01-27 by TuxSH