LumaTeam/Luma3DS #1581: Luma3DS pm does not gracefully fail when mounting SD

archshift, 2021-01-27:
Not sure exactly why, but the SD card may be configured such that FATFS (used by Luma and boot9strap) correctly mounts the partition, but the 3DS system is unable to.

Luma will correctly boot the system software, and the home screen will show "Could not detect an SD card." Upon trying to launch the gamecart, PM will crash at the following line: https://github.com/LumaTeam/Luma3DS/blob/ebeef7ab7f730ae35658b66ca97c5da9f663a17d/sysmodules/pm/source/termination.c#L14

This manifests as an _undefined instruction_ exception screen and it was completely non-obvious that the SD was the issue. Is there a way to display a more informative error message?

---

**SysNAND version (+emu/redNAND version if applicable):** 9.2.0

**Entrypoint (How/what you're using to boot Luma3DS):**

b9s

**Luma3DS version:**

18f3783f52dac9ca622cad44155d3bcd6af550ba

profi200, 2021-01-27:
This can also happen when HOS detects the write lock which is flaky on very beaten up SD card slots. It happened to me before ("Can't access the SD card blah" despite bare metal homebrew not having a problem).

TuxSH, 2021-01-27:
There's no way to gracefully handle this. There is no way to properly display the error message, and it would a similar assertion to fail down the road in the subsequent firmlaunch patches (that reload Luma, basically).

closed 2021-01-27 by TuxSH

archshift, 2021-01-29:
If Luma already has an error display screen, how is there no way to handle this? There's clearly some way to display errors already, as evidenced by the custom exception handler. Can't Luma handle a custom SVC that does the job?