LumaTeam/Luma3DS #1582: [Feaure Request] Autostart options like in Priiloader

DraconicNEO, 2021-01-27:
Basically the idea is to have a boot menu where you can select an app to automatically boot into bypassing the System menu, and also maybe an option to redirect apps returning to the home menu, although that might be harder. Ideally it should be able to autoboot Games/Apps in 3DS and TWL modes. I feel like autobooting GBA injects should also be allowed even though there's not much reason for it but it isn't needed, TWL autobooting is needed though as you won't be able to boot Twilight Menu without it.

Disclaimer(s):

- I'm **NOT** suggesting anything like Priiloader patches, that is way too much work and would probably get a lot of opposition for many different reasons.
- This is also Not a post about porting Priiloader to the 3DS, I don't know how someone could be so dumb as to want to port Wii homebrew to the 3DS.

profi200, 2021-01-27:
For GBA mode there is not much reason to use "injects" anymore other than for the pretty icon/banner because of open_agb_firm. DS(i) mode has not been re-implemented fully yet but someone else is working on that.

I don't know if this is worth the effort since it requires patches.

TuxSH, 2021-01-27:
I plan to add supporting autoloading via modifying firmlaunch parameters in a future release.

DraconicNEO, 2021-07-12:
Can someone add the enhancement tag to this?

Reaxt, 2023-04-04:
I would find this feature very useful still if it's not in already.