LumaTeam/Luma3DS #1586: Rhythm Heaven Megamix won't load, boots to black screen

INFINITE-PIKA, 2021-01-30:
**System model:**

New 2DS XL
**SysNAND version (+emu/redNAND version if applicable):**
11.14.0-46J Sys, no EMU
**Entrypoint (How/what you're using to boot Luma3DS):**
The Luma updater says I have a SIGHAX payload (I have boot9strap, though)
I hacked from seedminer comboed into UNSAFE_MODE

**Luma3DS version:**
luma3ds v10.2.1-d0a44fd2 (I think it's this https://github.com/Nanquitas/Luma3DS)
**Luma3DS configuration/options:**

Screen brightness: (4 )

Splash: (N/A )

Splash duration: ( N/A)

PIN lock: (N/A )

New 3DS CPU: (Normal/off)
Enable loading external FIRMs and modules: (Yes )

Enable game patching: (Yes)

Show NAND or user string in System Settings: (Yes )

Show GBA boot screen in patched AGB_FIRM: (Yes)

Set developer UNITINFO: (No)

Disable Arm11 exception handlers: (No)

Enable Rosalina on SAFE_FIRM: (No)

**Explanation of the issue:**
I'm using a Japanese Sysnand on an American sold system.
Whenever I try to boot Rhythm Heaven Megamix (the US version), after I get to the 3ds before-game splash thing, I get a black screen.
The JP version works, but I can't read all of the game, and would like to play in English if possible.
Even by using Luma Locale Changer to change the game to US, I still get a black screen.
I can boot to Rosalina but not the Home Menu.

**Steps to reproduce:**

1. Use JP sysnand, install cia of Rhythm Heaven Megamix

2. Get black screen, after the 3ds screen

No error dump. Sorry.

quadsYT, 2021-02-11:
Woah woah woah where did you get the cia from

INFINITE-PIKA, 2021-02-11:
> Woah woah woah where did you get the cia from

both are from hShop.

quadsYT, 2021-02-12:
just checked that and its a piracy tool

quadsYT, 2021-02-12:
no help for you

INFINITE-PIKA, 2021-02-12:
> no help for you

thank you for your kindness tho bro, i really appreciate it

INFINITE-PIKA, 2021-02-19:
I solved it by reinstalling and using Luma Locale Changer. Thanks a ton @quadsYT, you were such a big help

closed 2021-02-19 by INFINITE-PIKA