LumaTeam/Luma3DS #1589: Crash on Nanquitas' BootNTR Selector, Crash Dump Included

K-dex, 2021-02-04:
Hi, I'm consistently getting an error after choosing any NTR version to use in app.
Can anyone help?
I've completely wiped the SD and started again but still no changes.

[crash_dump_00000000.zip](https://github.com/LumaTeam/Luma3DS/files/5926179/crash_dump_00000000.zip)

urherenow, 2021-02-04:
No. This is Luma3ds' GitHub, not bootNTR.

K-dex, 2021-02-05:
I completely agree, but I was under the impression it might be an error in Luma. 
Obviously diagnosing an issue revolves around scrutinising every element that could go wrong, my first port of call being the Luma software.
I assumed someone here would be able to help me understand the crash dump.
I'm unfortunately one of those newbies who've recently followed the hacks.guide section of the internet, as such my intelligence on the matter is miniscule.
As it's late where I am, I'll try and bring it up in their GitHub in the morning.
Thank you

urherenow, 2021-02-05:
Please search/look through closed issues before making a new one. You aren't the first to do/ask this very question. Also, search gbatemp. That that place is more suited for help threads.

closed 2021-02-09 by TuxSH