LumaTeam/Luma3DS #1592: Hi guys, I have the same problem, but switching the update files didn't do anything.

tardorgi, 2021-02-09:
Hi guys, I have the same problem, but switching the update files didn't do anything.
I formatted my SD Card too, thinking it would help.
Does someone know how to fix this ?

I'm greatful for your ideas :)

_Originally posted by @Neo564 in https://github.com/LumaTeam/Luma3DS/issues/1537#issuecomment-745218891_

i made the same mistake, i deleted all files on my old 3ds XL, even with the lattest boot.firm on it, it only gets the same error as the original post that i got the comment, someone please help

tardorgi, 2021-02-09:
[crash_dump_00000001.zip](https://github.com/LumaTeam/Luma3DS/files/5947974/crash_dump_00000001.zip)
this is the crash dump, 
[image]
and this is all the files i have even when i insert the card and try to turn it on

TuxSH, 2021-02-09:
You did not properly update.

Make sure Luma3DS is booted from the SD card: press SELECT at boot, look at what's written in yellow on the bottom screen bottom-left corner. If it says "booted from NAND" then there is a problem with your SD card (is it formatted as FAT32?)

tardorgi, 2021-02-09:
> You did not properly update.
> 
> Make sure Luma3DS is booted from the SD card: press SELECT at boot, look at what's written in yellow on the bottom screen bottom-left corner. If it says "booted from NAND" then there is a problem with your SD card (is it formatted as FAT32?)

it says booted from SD via FIRM0, my card is formatted as fat 32

Neo564, 2021-02-09:
> You did not properly update.
> 
> Make sure Luma3DS is booted from the SD card: press SELECT at boot, look at what's written in yellow on the bottom screen bottom-left corner. If it says "booted from NAND" then there is a problem with your SD card (is it formatted as FAT32?)

I've put the 2 files in the root and tried to boot as you said and it still has the same Error.
Yellow writing at the bottom says: Booted from SD via B9S

Ralkage, 2021-02-10:
> I've put the 2 files in the root and tried to boot as you said and it still has the same Error.
> Yellow writing at the bottom says: Booted from SD via B9S

Same here; I'm not sure what happened but I was playing an old DS cart and when I woke up this morning to start up my 3DS, I got the same error as OP.

tardorgi, 2021-02-21:
Well, i tried a lot of things, using older versions of luma, trying game cartridges, formating sd card to exfat, fat 32, etc, nothing works, any new solution?

eric100834, 2021-02-27:
If your not worried about WiFi turn it off I was getting same error but when I turned WiFi off it stopped. As long as WiFi is on mine does it.. my problem is I have 11.3 and there software supports 11.14 if yours is 11.3 that’s why the error is coming up. Turn WiFi off when not needing it

Shadow322, 2021-02-27:
> If your not worried about WiFi turn it off I was getting same error but when I turned WiFi off it stopped. As long as WiFi is on mine does it.. my problem is I have 11.3 and there software supports 11.14 if yours is 11.3 that’s why the error is coming up. Turn WiFi off when not needing it

I'm why arnt u on 11.14?

tardorgi, 2021-03-21:
> If your not worried about WiFi turn it off I was getting same error but when I turned WiFi off it stopped. As long as WiFi is on mine does it.. my problem is I have 11.3 and there software supports 11.14 if yours is 11.3 that’s why the error is coming up. Turn WiFi off when not needing it

i tried this, but i can't turn on or off the WiFi, the yellow light don't turn on no matter what i do

jack22128, 2021-04-10:
I am having a very similar problem where I have the same error screen but my 3ds keeps booting from CTRNAND via B9S and if I take the sd card out it is still doing it, it was working earlier but since I transfered my files to my pc and back again it won't boot from the sd card and I keep getting the error screen. (Power and start doesn't work for godmode 9 anymore) The luma 3ds version it keeps booting is 10.0.1 when I have 10.12.1 installed on the root of my sd card. Any fixed?

urherenow, 2021-04-11:
You don't need an sad card to boot from ctrnand. That's the whole purpose of adding Luma to it. Your 3ds can't read your card. Either it's not formatted properly, or your sd slot is damaged.

TuxSH, 2021-07-16:
Reinstall b9s properly using SafeB9SInstaller

closed 2021-07-16 by TuxSH