LumaTeam/Luma3DS #1596: Luma claims the system is booted with B9S when using fastboot3DS

itstoast27, 2021-02-16:
**System model:**

[Old 2DS]

**SysNAND version (+emu/redNAND version if applicable):**

[11.13.0-45U SysNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[fastboot3DS]

**Luma3DS version:**

[https://github.com/Nanquitas/Luma3DS/releases/tag/v10.2.1, also apparent on Stable v10.2.1]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4 )

Splash: ( Off)

Splash duration: ( 1)

PIN lock: ( off )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( x )

Show NAND or user string in System Settings: ( x )

Show GBA boot screen in patched AGB_FIRM: ( x )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
The yellow text at the bottom states how and with what Luma was booted with, and when using fastboot3DS, Luma claims to be booted with B9S.

**Steps to reproduce:**

1. Use fastboot3DS.

2. Boot into Luma's config menu.

**Dump file:**

profi200, 2021-02-16:
This is not Luma's fault. It expects to be booted in a b9s like environment which fb3DS mimics. Can't really detect if it really was b9s or fb3DS.

itstoast27, 2021-02-16:
Would it be possible to have a flag of sorts within B9S/fb3DS that Luma picks up on, and uses to tell which one it was booted with? I know it's minor, but I wanna see it happen for some reason haha
Should I make an issue on the fb3DS and B9S repos for the incorporation of this?

profi200, 2021-02-16:
b9s passes a number on to Luma and co. that identifies it. We use the same for compatibility reasons in fb3DS. Unlikely either side will change anything to detect anything but b9s.

closed 2021-07-19 by TuxSH