LumaTeam/Luma3DS #1603: Dragon Quest 8 Non-Functional

Alibrooks, 2021-03-03:
The feature of removing region locking for physical games is non-functional for Dragon Quest 8 (NTSC console, PAL game). Every single solution for this has not worked. I am on a n3DSXL ver. 11.14.0, Luma ver. 9.1

There is no option to import the seed, Luma Locale Switcher does not work (even with the correct locales directory), and manually making a txt file in Luma > titles did not work. The game boots up before saying "This title cannot be played without extra data." Before trying to create the extra data and failing a few seconds in citing "An error occurred while the SD card was being written to. Please turn off the system and re-insert the SD card."

Shadow322, 2021-03-03:
> The feature of removing region locking for physical games is non-functional for Dragon Quest 8 (NTSC console, PAL game). Every single solution for this has not worked. I am on a n3DSXL ver. 11.14.0, [...]

Update your Luma3DS

Alibrooks, 2021-03-03:
> Update your Luma3DS

I did that before all of this, and whenever I click on the updater all I get is "Could not obtain update details." and I'm on the internet

Shadow322, 2021-03-03:
> > Update your Luma3DS
> 
> I did that before all of this, and whenever I click on the updater all I get is "Could not obtain update details." and I'm on the internet

https://github.com/LumaTeam/Luma3DS/releases/download/v10.2.1/Luma3DSv10.2.1.zip

Download this extract the file place boot.firm on root of SD card then boot your 3ds

closed 2021-07-16 by TuxSH