LumaTeam/Luma3DS #1604: Feature Request: Stealth mode.

zaphod77, 2021-03-04:
This would be a new configuration option.

If Stealth mode is turned on. it would have the following effects.

1) select to bring up configuration disabled.
2) region free disabled. (to prevent running an out of region cart to detect CFW)
3) chainloader disabled
4) rosalina hotkeys disabled.
5) pin lock disabled.
6) version string change disabled.
7) homebrew app replacement disabled.
8) anything else that could be used to determine that the console has CFW disabled.

The main point of this is for booting in sysnand, and keeping the config file there for the sysnand version.  Say you want to go play Pokemon at an event.

Bu injecting luma to ctrnand, then booting with select and enabling stealth mode, and putting in a clean sdcard, it becomes impossible for any nosy judge to detect that the console was hacked, and disqualify you, provided nothing non legit is installed to twl_nand or the sdcard currently in. ( if a judge sees homebrew launcher, or pksm or  fbi or something on your home screen, well...)

If your firmware is on sdcard, you can simply remove the config file to restore normal operation.  But the point of this is to keep a clean sdcard and hide all evidence of the hack. 

The judge requesting hotkey presses will reveal nothing.  Even if they make you remove your sdcard, and look at it themselves, and boot your console without it, they will see nothing. :)

You can always boot an sdcard and go into godmode 9 to change things on the CTRnand version.

zaphod77, 2021-03-04:
not actually a rosalina feature. it's a new config screen setting. so i made it here.

xfix, 2021-03-10:
I'm not sure what you are asking for. Official Pokemon events for sixth and seventh generation games are pretty much gone, and I doubt that anyone other than Nintendo would care about 3DS having custom firmware installed.

Datsami000, 2021-03-18:
I think this would be a cool idea, just highly doubt it would do much in 2021. This might be a bit too little, too late.

zaphod77, 2021-03-18:
Yeah.  if there's truly no more events for 3ds games, there's no reason to
bother.  Would be a nice trick, though. I wasn't sure of that.

TuxSH, 2021-07-19:
For most things, editing `sysmodules/Makefile` should be sufficient.

closed 2021-07-19 by TuxSH