LumaTeam/Luma3DS #1609: Input Redirection in DS Mode

zs2212221, 2021-03-16:
This is a pretty general question, but what would it take to implement input redirection when launching a game in DS mode? Is this even something at the Luma3DS-level?

Shadow322, 2021-03-16:
> This is a pretty general question, but what would it take to implement input redirection when launching a game in DS mode? Is this even something at the Luma3DS-level?

Not possible at all

zs2212221, 2021-03-16:
Is this because Luma3DS CFW and since DS games launch in DS mode, Luma literally can't do anything about that?

Shadow322, 2021-03-16:
> Is this because Luma3DS CFW and since DS games launch in DS mode, Luma literally can't do anything about that?

Yep

profi200, 2021-03-16:
I would not say this is impossible but not doable with Luma/Rosalina. A custom TWL_FIRM is needed. And you will run into issues when the game tries to use WiFi.

TuxSH, 2021-07-16:
It's like profi said, and a ginormous PITA in any case.

closed 2021-07-16 by TuxSH