LumaTeam/Luma3DS #1610: I have an issue here that need support. So I bought a new sd card for my cfw new 3ds xl, because I wanted to upgrade my storage. I formatted the sd card to FAT32..something and transferred all the files from the old sd card to the new one by copying all the files to the pc and transferred it to the new sd card. After that I noticed that if I go to my games and cfw apps the top screen is blank it the title of the games or apps don't appear and when I try to open them an error would occur it said "An error occurred (ErrDisp), Error type: card removed, the card was removed, press any button to reboot", then it would just reboot and I cant play my games or open some cfw applications. By the way I bought this 3ds already hacked and I didn't actually hacked the 3ds on my own but if you guys can give me instructions I'll be fine.

UsernameJV, 2021-03-16:
**System model:**

[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

urherenow, 2021-03-16:
GitHub isn't a support forum. You didn't fill in a single part of the template either. The only useful information you provided was "new card" and "new 3ds xl".

Chances are, you bought a fake card, or it's faulty. At any rate, you should be asking for help on discord or gbatemp.

UsernameJV, 2021-03-17:
Thanks for the support and sorry for the bother.

closed 2021-07-16 by TuxSH