LumaTeam/Luma3DS #1614: Cannot download updates or Official themes, (ErrDisp)

MewtwoFanz, 2021-03-23:
I have been having this issue for awhile now, and I can't install updates nor official themes. This is what i get in my error:

An error occurred (ErrDisp)

Error type:    generic

Process ID:              19
Process name:         am
Process Title ID:       0x0004013000001502

Address:                  0x001093aa
Error Code:              0xd900182f

urherenow, 2021-03-23:
Are you seriously trying to update online and even downloading official things from the eshop, while running CFW?

Try in OFW and see if you're banned.

Otherwise, there is a ton of info you're required to provide that you haven't. Like which firmware, system model, and Luma version you're currently running. And whether you're using b9s, fastboot3ds, a9lh, etc

MewtwoFanz, 2021-03-23:
I meant game updates, not system updates. I should have clarified. My bad. I can still do system updates.

MewtwoFanz, 2021-03-23:
I am running the latest firmware, 11.14, n3DS xl. I still need to look for the rest of the settings. I'll update this once I have them.

Dimensional, 2021-03-23:
If your system is banned from online services, the only updates you can receive are system updates. Anything that requires downloading from the eShop, such as DLC, Game Updates, or Themes, will be blocked. I'm in the same boat with the same issues, and there is no way Luma can unban your system for these.

MewtwoFanz, 2021-03-23:
I am not banned. These updates were working just fine before this problem occured.

urherenow, 2021-03-23:
Can you open and browse the eshop?

Dimensional, 2021-03-23:
> 
> 
> I am not banned. These updates were working just fine before this problem occured.

If you have "Use developer UNITINFO" enabled, disable it and try again. If you receive an error code with a message like "access to online services has been restricted by Nintendo", then your system has been banned. Don't just complain about it being the CFWs fault until everything else has been ruled out please, especially when there has been no information provided to rule it out. There is a very real chance that is the case.

Also to determine if it's a generic error, try disabling "ARM11 exception handlers", which is also found in the boot settings for Luma. This is only to get a better idea of what the problem is.

MewtwoFanz, 2021-03-23:
I am able to enter the eshop, and my luma version is v10.2.1-4ae4d16d configuration. But it didnt matter what version I used, I still got the same error, even on the latest luma version.

MewtwoFanz, 2021-03-23:
If this is important, I also have gm9, hourglass9, and decrypt9wip running

MewtwoFanz, 2021-03-23:
I don't have disable Arm11 exception handlers enabled. It's always been disabled

MewtwoFanz, 2021-03-23:
Set developer UNITINFO is also disabled

MewtwoFanz, 2021-03-23:
I just tried it on luma v10.2.1-d0a44fd2-3gxldr configuration, still same error.

MewtwoFanz, 2021-03-23:
The 3gx is the version i want to use

closed 2022-06-05 by suppai