LumaTeam/Luma3DS #1615: he won't let me play tomodachi life

CamilaSHP, 2021-03-30:
I have a bug when I want to install the tomodachi life game, it lets me install it and everything but when I try to enter I get a problem with the luma, this comes up.

an expection ocurred

processor:                     arm11(core 0)
exception:                     prefetch abort (svcBreak)
fault status:                  debug event
current process:          app(0004000000008c400)

R0            00000000              R1             00000000
R2            00000000              R3             00A7CA08
R4            081B69D0             R5              FFFFFF80
R6            00A7C900             R7              00000000
R8            0FFFFE28              R9              00000000
R10          00000000             R11             00000000
R12          00000000             SP               0FFFFE28
LR            00132B3C             PC              0018F1D8
CPSR      60000010              FPEXC       40000700   

You can find the dump in the following file
SD: /luma/dumps/arm11/crash_dump_00000009.dmp

Press any button to shutdown

(IN THE SCREEN BELOW THIS IS OUT)

Stack dump

0FFFFE28 0C 78 A4 00 00 00 00 00
0FFFFE30 00 00 00 00 00 00 00 00
0FFFFE38 00 00 00 00 00 00 00 00
0FFFFE40 D0 FE FF 0F 60 00 00 00
0FFFFE48 D0 69 1B 08 40 D2 16 08
0FFFFE50 4E 04 00 00 D0 FE FF OF
0FFFFE58 00 C9 A7 00 DC 3D 16 00
0FFFFE60 40 D2 16 08 26 00 00 00
0FFFFE68 4E 04 00 00 D0 FE FF OF
0FFFFE70 00 C9 A7 00 00 00 00 00
0FFFFE78 00 00 00 00 00 00 00 00
0FFFFE80 20 6A 1B 08 5C 35 18 00
0FFFFE88 00 C9 A7 00 08 00 00 00
0FFFFE90 00 00 00 00 87 01 10 00
0FFFFE98 00 00 00 00 00 00 00 00
0FFFFEA0 E0 60 21 08 00 00 00 00
0FFFFEA8 80 C5 AC 00 5C A6 A7 00
0FFFFEB0 00 00 00 00 10 A6 39 00
0FFFFEB8 80 00 00 00 00 00 00 00

PS: The only thing I can think of is to download another version of luma but I don't know which one and if I do, I don't know if the other games I have will continue to work.

PD2: It is with the only game that I had this type of error.

PD3: My console is the new 2ds xl it is updated to version 11.14.0

PD4: I have tried several companies to see if it was that error but with all of them I get the same message

Han121519, 2021-04-22:
mine too :'(

Oreo639, 2021-04-22:
Is this a cartrige game or digital?

CamilaSHP, 2021-04-23:
https://www.youtube.com/watch?v=gdlxADJhnXU&t=1s&ab_channel=DarkEvil

This video made the game work on my console, I don't know if it works for
you, the video has lyrics in Spanish

pd:If you need help I can translate what the video says into English

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Oreo639, 2021-04-23:
This isn't a luma issue. If you have a cartridge copy, you can try redumping it, otherwise, piracy isn't supported.

Han121519, 2021-04-24:
you have to download "luma locale switcher" from "universal updater" and if you finished downloading "luma locale switcher" launch it and there has 3 chooses (titles, choose locales directory, and reset all to system default) choose "choose locale directory" open it and select "/luma/titles/%s/locale.txt" and wait it and when you return to main menu choose titles and select your game to change and set up the region and language that should be (region:usa language:english)

closed 2021-07-16 by TuxSH

Themis3000, 2022-06-11:
Adding onto @Han121519's fix

You may also need to enable file patching in the luma3ds menu

Turn off the console, hold select and power the device on, then enable "file patching" (it's named something like that) and then press start to save and press again to get into godmode9, once there press start again to reboot. Should work after that!

Many thanks to u/timchenw for posting this fix on reddit originally