LumaTeam/Luma3DS #1616: [FEATURE REQUEST] Audio volume and style chooser in Rosalina menu

Mr7011cfw, 2021-03-30:
You would be able to change volume and the audio from mono to stereo to surround much quicker and easier if you need to or can't.

Mewthree3, 2021-04-01:
Does the 3DS even have volume (mono/stereo/surround) options?

Mr7011cfw, 2021-04-01:
Yes, in the settings and or music app you can choose between mono, stereo,
and surround sound. Being able to set the audio volume would be more
helpful but being able to change the style right from the rosalina menu
would also be faster and easier to access.

Mewthree3, 2021-04-01:
I see! I do think that would be cool. Especially a volume slider...

Mr7011cfw, 2021-04-01:
Rad😎

junpeikun, 2021-07-26:
Also, I don't know if I'm the only one experiencing this, but my volume slider is VERY sensitive, to the point where just holding my 3DS can trigger a volume change. A way to completely disable that slider and replace it by a software counterpart is a must.

Mewthree3, 2021-07-26:
If I remember correctly, there’s a Luma3DS fork that can solve this for you. I forget what it’s called as I don’t have my 3DS on hand, but it’s avaliable on Universal Updater!

suppai, 2022-06-05:
Duplicate: #1485

closed 2022-06-05 by suppai