LumaTeam/Luma3DS #1619: Request: Functionality to have a cheatcode enabled automatically?

Clydefrosch22, 2021-04-04:
There are certain cheats that I'd like to be just always active from the moment I start a game, like increased walking speed or exp multipliers in some rpgs.
As far as I can tell, cheats are always off at the start of a game and I wonder if it's possible to add some indicator so luma knows 'This cheat is always supposed to be on'.
Like something to add into the cheat file, like (X) or *** after or at the start/end of the codes name

[Max Money]***
[*Max Money*]
Auto:[Max Money]

or something of that sort. Of course, it would be even better if that could be set within the Luma Cheat Menu too.

Low priority request for sure, but I think it would be a nice quality of life addition.

Mewthree3, 2021-04-06:
I'd love this too, there are some cheats that I always want on lol

devlman127, 2024-07-01:
I'd also love this feature, hope this gets a look at

riggles1, 2025-03-23:
This really needs to happen, things like 60fps codes or the controls fix for A Link Between Worlds (fixes the circle pad angle, as it's otherwise 5 degrees off)