LumaTeam/Luma3DS #1620: Arm11 Core 0 issue with Both Metroid 3ds games

DaftDisc, 2021-04-07:
**System model:**
New 3DS XL
 
**SysNAND version (+emu/redNAND version if applicable):**

11.14.0-46J

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

10.2.1

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
I bought a new copy of both Metroid: Samus Returns and Metroid prime: Federation force. Upon opening them and trying to play them I was met with this screen that from further investigation is very similar to a screen someone reported from ACNL
in which they replaced a gardens thing but I have no clue how or what is the actual cause of this. For information sake I am using a japanese 3ds xl with a us game cart. (All other games and game carts work)

[image]
[crash_dump_00000000.zip](https://github.com/LumaTeam/Luma3DS/files/6268819/crash_dump_00000000.zip)

**Steps to reproduce:**

1. Try to launch any copy (dumped, or game cart)

**Dump file:**

peterstory, 2021-05-15:
For what it's worth, I've gotten the same error on my New 3DS LL when playing US region games. I can't remember what game it was – probably Bravely Default. Usually I get the error after a few minutes of playing.

IAmBrenna, 2021-05-23:
I got a similar error with Old 3DS.

**Steps to reproduce**
Launch Pokemon ultra moon and enable 2 cheats from Rosalia menu.
After a battle the game crashes.

[crash_dump_00000000.zip](https://github.com/LumaTeam/Luma3DS/files/6528427/crash_dump_00000000.zip)

closed 2022-06-05 by suppai