LumaTeam/Luma3DS #1625: An exception occurred Arm 11, I accidently erased everything from my SD card including all my custom firmware files,  once I inserted it I tried fixing it with a new files that should fix it but due to me not having any of my custom firmware files I can't possibly do it.

EnfernoToad, 2021-04-13:
**System model:**

[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

urherenow, 2021-04-13:
You don't know how hard it is to not just crush your soul into the dirt... or do you?

You post an "issue" without filling in one iota of the template.

You say there's no way you can possibly get the firmware files... while posting this in the very site you're supposed to get the files from.

I'm at a loss for words.

EnfernoToad, 2021-04-13:
> You don't know how hard it is to not just crush your soul into the dirt... or do you?
> 
> You post an "issue" without filling in one iota of the template.
> 
> You say your issue is not having "the firmware files", while posting this in the very site you're supposed to get the files from.
> 
> I'm at a loss for words.

I meant since I removed my sd card and erased everything because I forgot about my 2ds xl which has custom firmware that was running with the same SD card and even tho I downloaded and copied those files to the root of my SD card it still didn't work also you can go to hell u bastard

radostin04, 2021-04-15:
what the absolute fuck is this issue

Mewthree3, 2021-04-19:
@EnfernoToad Do you have B9S (Boot9Strap) installed? (It stays on even if you don't have an SD card in)
I assume that since you formatted your SD card accidentally, you wouldn't have a backup, right?
I know that if your DS is bricked (which I think you're describing), you usually need a backup to restore, unless if you're able to boot into recovery mode.
So, could you tell me if booting into recovery mode worked? (Hold UP+A+L+R when you power on)

closed 2021-07-16 by TuxSH