LumaTeam/Luma3DS #1629: How do I fix ErrDisp Error whenever I play a ds game

Reshiram76, 2021-04-23:


Shadow322, 2021-04-23:
Using a ds flashcart?

Reshiram76, 2021-04-23:
> Using a ds flashcart?

Nope, legit copys

Reshiram76, 2021-04-26:
here is my error code

Error type:       generic

Process ID:       19
Process name:     am
Process title ID: 0x0004013000001502

Address:          0x001093aa
Error code:       0xc8804478

Shadow322, 2021-04-26:
> here is my error code
> 
> Error type: generic
> 
> Process ID: 19
> Process name: am
> Process title ID: 0x0004013000001502
> 
> Address: 0x001093aa
> Error code: 0xc8804478

The error code is generic

Reshiram76, 2021-04-26:
what do I do about that

urherenow, 2021-04-26:
A good start would be to use the bug template, filling in all of the info, so we at least know where you're coming from. Also, have you ever installed a patched twl_nand (also known as twl_firm)? Whether you have or not, you should try to obtain an unpatched copy and install it, before troubleshooting this issue.

Reshiram76, 2021-04-26:
So what your saying is that I should find one and install it then do a system update?

Reshiram76, 2021-04-26:
> A good start would be to use the bug template, filling in all of the info, so we at least know where you're coming from. Also, have you ever installed a patched twl_nand (also known as twl_firm)? Whether you have or not, you should try to obtain an unpatched copy and install it, before troubleshooting this issue. Just make sure you grab the correct one (it's different between o3ds and n3ds, for example...)
> 
> If you're sure you know the correct title, you could even delete it with FBI, then run a system update, to get an unpatched one installed.

FIXED, PROBLEM SOLVED THANKS.

closed 2021-04-26 by Reshiram76