LumaTeam/Luma3DS #1630: issue when compiling luma3ds from git

silverhikari, 2021-05-01:
i am compiling Luma3ds on manjaro, i have made and installed makerom, i have also install firmtool as stated in the Readme. i also have the latest version of devkit arm(as of the moment) and libctru(version is 2.0.1-1).
here is the error report

https://paste.ee/p/p4oAs

Oreo639, 2021-05-01:
Why do you need to compile Luma3DS from git?

Shadow322, 2021-05-01:
> Why do you need to compile Luma3DS from git?

He wants the latest compiled code

Oreo639, 2021-05-01:
If that *literally* all you want, wouldn't you save yourself some time by using the [latest release](https://github.com/LumaTeam/Luma3DS/releases) or an [hourly](https://github.com/hax0kartik/luma-hourlies/releases)?

silverhikari, 2021-05-01:
did not know the hourly existed so thanks that looks like what i wanted.

urherenow, 2021-05-01:
Just as the main page says, you need the latest libctru. In this case, including commits not released yet.

So clone libctru, build it, then "sudo -HE make install"

Lust built the latest Luma3ds a few seconds ago with no errors (not that it doesn't have any, but the latest commit disabled errors that weren't important, according to the devs...)

closed 2021-07-16 by TuxSH