LumaTeam/Luma3DS #1633: Cannot boot Miitopia Cart

SuperKirbylover, 2021-05-19:
**System model:** New 3DS XL
**SysNAND version (+emu/redNAND version if applicable):** Sys 11.13.0-45U
**Entrypoint (How/what you're using to boot Luma3DS):** Boot9Strap
**Luma3DS version:** v10.2.1 stable

**Luma3DS configuration/options:**

Screen brightness: ( 4 )
Splash: ( Before )
Splash duration: ( 3 )
PIN lock: ( Off )
New 3DS CPU: ( Off )

--

Enable loading external FIRMs and modules: ( )
Enable game patching: ( X )
Show NAND or user string in System Settings: ( X )
Show GBA boot screen in patched AGB_FIRM: ( )
Set developer UNITINFO: ( )
Disable Arm11 exception handlers: ( )
Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
I booted up Miitopia, and before the splash screen for the 3DS fades out, I encounter this crash. It's usually the same error. Says its a Arm11 Core 1 issue, with the exception type undefined instruction. Current process is loader, which is nothing but 0's.

**Steps to reproduce:**
1. Have a copy of Miitopia for the Nintendo 3DS inserted into the New Nintendo 3DS XL
2. Boot it

**Dump file:**
[crash_dump_00000001.zip](https://github.com/LumaTeam/Luma3DS/files/6505498/crash_dump_00000001.zip)

Mewthree3, 2021-05-21:
Try to dump your game file using [https://github.com/aspargas2/CartInstall](url), and see if you're able to play it like that. 
It sounds weird but I have a feeling it could help.

wg481, 2021-05-31:
That seems really odd. I have had no trouble booting up gamecards on Luma. Honestly, I would redo the whole Luma process.
Frankly, it sounds like a system issue. If you can, you could try booting up the card on your 3DS Luma-less (if you don't have illegitimate CIAs) and see what happens.

closed 2022-06-05 by suppai