LumaTeam/Luma3DS #1635: I can't load my homebrew mods because i get a crash from luma: Processor: Arm11 ( core 0 ) Exception type: prefetch abort Fault status: Translation - Section Current process: SKATER ( 0004003000009402 )

DerekEpicGamer, 2021-06-06:
I keep on geting this error mesage from luma every time i boot up homebrew.
[image]

can you guys help me fix it?

Oreo639, 2021-06-06:
You don't need to use browserhax with luma.
You can follow this to launch the homebrew launcher with Rosilina:
https://3ds.hacks.guide/finalizing-setup

DerekEpicGamer, 2021-06-06:
> You don't need to use browserhax with luma.
> You can follow this (specifically section 3) to launch the homebrew launcher with Rosalina:
> https://3ds.hacks.guide/finalizing-setup
The menu is in chinese/japanise

Oreo639, 2021-06-06:
> > You don't need to use browserhax with luma.
> > You can follow this (specifically section 3) to launch the homebrew launcher with Rosalina:
> > https://3ds.hacks.guide/finalizing-setup
> 
> The menu is in chinese/japanise

What menu?

DerekEpicGamer, 2021-06-06:
the Rosalina Menu for download play

Oreo639, 2021-06-06:
That is likely an unofficial version of luma. If you redownload the latest release it should be in english.

DerekEpicGamer, 2021-06-07:
it works thanks alot

closed 2021-06-07 by DerekEpicGamer

justriy, 2022-05-20:
I had attempted the same thing as oreo639 had suggested but it still crashed with the same error but this time on core 1. I'm not sure if I deleted important files (i was removing excess files from the 3ds folder that I didn't need before this) or if I have to reinstall but anything that could help would be appreciated. (Image for reference)
[image]