LumaTeam/Luma3DS #1638: Attribution missing for source taken from Stack Overflow

gnif, 2021-06-14:
The following code has been copied from my answer on Stack Overflow which I am the original author of. This was posted under CC by-sa 4.0 with attribution required. Please alter your source to include such attribution or remove the offending code from your repository.

https://github.com/LumaTeam/Luma3DS/blob/99a2b8e3b2077d3b1d77087cacbd5e05de88da0c/sysmodules/rosalina/source/menus.c#L409-L448

Original source

https://stackoverflow.com/questions/21593692/convert-unix-timestamp-to-date-without-system-libs

TuxSH, 2021-06-14:
Sorry about that, looks like it slipped during the review of 8f03234.

closed 2021-06-14 by TuxSH

gnif, 2021-06-14:
No worries, thanks :)

commit 8e9a3bfc (2021-06-14): Fix #1638