LumaTeam/Luma3DS #1639: Gyro/Motion Control Calibration causes system to freeze

hydrated-3ds, 2021-06-14:
System model: New 3DS
SysNAND version (+emu/redNAND version if applicable): Sys 11.14.0-46U
Entrypoint (How/what you're using to boot Luma3DS): Boot9Strap
Luma3DS version: v10.2.1 3GX, [Build](https://github.com/Nanquitas/Luma3DS/releases/tag/v10.2.1)

Luma3DS configuration/options:

Screen brightness: ( 4 )
Splash: ( Off )
Splash duration: ( 1 )
PIN lock: ( Off )
New 3DS CPU: ( Off )

--

Enable loading external FIRMs and modules: ( X )
Enable game patching: ( X )
Show NAND or user string in System Settings: ( X )
Show GBA boot screen in patched AGB_FIRM: ( )
Set developer UNITINFO: ( )
Disable Arm11 exception handlers: ( )
Enable Rosalina on SAFE_FIRM: ( )

--

--

**Explanation of the issue:**

Trying to calibrate the motion sensor/gyroscope causes the system to freeze. Resetting to the default works, but does not fix my inverted motion controls.

**Steps to reproduce:**

1. Hold Y+B at the same time for a few seconds on the home menu with all games/applications closed.

2. Select Calibrate, then tap OK.

AuroraWright, 2021-06-14:
Can't reproduce on my New 3DS XL. Also please don't report issues with heavily modified forks like the one you mentioned, try again with the official release

hydrated-3ds, 2021-06-15:
> Can't reproduce on my New 3DS XL. Also please don't report issues with heavily modified forks like the one you mentioned, try again with the official release

Tried again with the official release and default settings, same issue. Might just be my system.

hydrated-3ds, 2021-06-15:
Tried again, worked this time for some reason.

closed 2021-06-15 by hydrated-3ds

MADOK1, 2022-08-08:
I'm having this problem on my old 3ds too.

Arnedeklerk, 2025-05-05:
Same issue here on old 3ds.