LumaTeam/Luma3DS #1640: Checkpoint isn’t working and I get an error

DerekEpicGamer, 2021-06-18:


sergioinfosec, 2021-07-07:
Hello, how are you today?

Let me try to help you on that. Do these steps.

- Power Off console
- Press select button + power button together
- Take a picture of the next screen, it's your Luma configs (don't worry)
- Press START to leave that screen
- Post the picture here

On your Luma Config will have a option called "Arm11 handlers exceptions", try to mark that option and check if the error keep happening.

I'm waiting for your test. See you!

eip618, 2021-07-07:
> Hello, how are you today?
>
> Let me try to help you on that. Do these steps.
>
> * Power Off console
> * Press select button + power button together
> * Take a picture of the next screen, it's your Luma [...]

1. This is an issue tracker, not a support forum.
2. Disabling arm11 exception handlers is, in 98% of cases, not going to solve the problem and will instead make things more difficult to diagnose.

sergioinfosec, 2021-07-13:
Thats true! Sorry about that. Another guy explained me about this function and really doesn't help. 

Please, ignore my last post. ^^

TuxSH, 2021-07-19:
Make sure that your Checkpoint version is up to date. Using 3dsx is recommended.

Otherwise this looks like a Checkpoint issue

closed 2021-07-19 by TuxSH