LumaTeam/Luma3DS #1642: Build Error

Midnoclose, 2021-06-19:
```
make[1]: Entering directory '/home/midnoclose/Development/Luma3DS/sysmodules'
make[2]: Entering directory '/home/midnoclose/Development/Luma3DS/sysmodules/loader'
make[3]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/midnoclose/Development/Luma3DS/sysmodules/loader'
make[2]: Entering directory '/home/midnoclose/Development/Luma3DS/sysmodules/sm'
make[3]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/midnoclose/Development/Luma3DS/sysmodules/sm'
make[2]: Entering directory '/home/midnoclose/Development/Luma3DS/sysmodules/pm'
make[3]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/midnoclose/Development/Luma3DS/sysmodules/pm'
make[2]: Entering directory '/home/midnoclose/Development/Luma3DS/sysmodules/pxi'
make[3]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/midnoclose/Development/Luma3DS/sysmodules/pxi'
make[2]: Entering directory '/home/midnoclose/Development/Luma3DS/sysmodules/rosalina'
menus.c
/home/midnoclose/Development/Luma3DS/sysmodules/rosalina/source/menus.c: In function 'RosalinaMenu_ShowDebugInfo':
/home/midnoclose/Development/Luma3DS/sysmodules/rosalina/source/menus.c:89:5: error: unknown type name 'FS_SdMmcSpeedInfo'
   89 |     FS_SdMmcSpeedInfo speedInfo;
      |     ^~~~~~~~~~~~~~~~~
In file included from /opt/devkitpro/libctru/include/3ds.h:13,
                 from /home/midnoclose/Development/Luma3DS/sysmodules/rosalina/source/menus.c:27:
/home/midnoclose/Development/Luma3DS/sysmodules/rosalina/source/menus.c:110:49: error: passing argument 1 of 'FSUSER_GetSdmcSpeedInfo' from incompatible pointer type [-Werror=incompatible-pointer-types]
  110 |         if (R_SUCCEEDED(FSUSER_GetSdmcSpeedInfo(&speedInfo)))
      |                                                 ^~~~~~~~~~
      |                                                 |
      |                                                 int *
/opt/devkitpro/libctru/include/3ds/result.h:9:30: note: in definition of macro 'R_SUCCEEDED'
    9 | #define R_SUCCEEDED(res)   ((res)>=0)
      |                              ^~~
In file included from /opt/devkitpro/libctru/include/3ds/services/am.h:7,
                 from /opt/devkitpro/libctru/include/3ds.h:33,
                 from /home/midnoclose/Development/Luma3DS/sysmodules/rosalina/source/menus.c:27:
/opt/devkitpro/libctru/include/3ds/services/fs.h:470:37: note: expected 'u32 *' {aka 'long unsigned int *'} but argument is of type 'int *'
  470 | Result FSUSER_GetSdmcSpeedInfo(u32 *speedInfo);
      |                                ~~~~~^~~~~~~~~
/home/midnoclose/Development/Luma3DS/sysmodules/rosalina/source/menus.c:112:46: error: request for member 'sdClkCtrl' in something not a structure or union
  112 |             u32 clkDiv = 1 << (1 + (speedInfo.sdClkCtrl & 0xFF));
      |                                              ^
/home/midnoclose/Development/Luma3DS/sysmodules/rosalina/source/menus.c:115:31: error: request for member 'highSpeedModeEnabled' in something not a structure or union
  115 |                 (int)speedInfo.highSpeedModeEnabled, SYSCLOCK_SDMMC / (1000 * clkDiv)
      |                               ^
/home/midnoclose/Development/Luma3DS/sysmodules/rosalina/source/menus.c:115:54: error: 'SYSCLOCK_SDMMC' undeclared (first use in this function); did you mean 'SYSCLOCK_SOC'?
  115 |                 (int)speedInfo.highSpeedModeEnabled, SYSCLOCK_SDMMC / (1000 * clkDiv)
      |                                                      ^~~~~~~~~~~~~~
      |                                                      SYSCLOCK_SOC
/home/midnoclose/Development/Luma3DS/sysmodules/rosalina/source/menus.c:115:54: note: each undeclared identifier is reported only once for each function it appears in
In file included from /opt/devkitpro/libctru/include/3ds.h:13,
                 from /home/midnoclose/Development/Luma3DS/sysmodules/rosalina/source/menus.c:27:
/home/midnoclose/Development/Luma3DS/sysmodules/rosalina/source/menus.c:118:49: error: passing argument 1 of 'FSUSER_GetNandSpeedInfo' from incompatible pointer type [-Werror=incompatible-pointer-types]
  118 |         if (R_SUCCEEDED(FSUSER_GetNandSpeedInfo(&speedInfo)))
      |                                                 ^~~~~~~~~~
      |                                                 |
      |                                                 int *
/opt/devkitpro/libctru/include/3ds/result.h:9:30: note: in definition of macro 'R_SUCCEEDED'
    9 | #define R_SUCCEEDED(res)   ((res)>=0)
      |                              ^~~
In file included from /opt/devkitpro/libctru/include/3ds/services/am.h:7,
                 from /opt/devkitpro/libctru/include/3ds.h:33,
                 from /home/midnoclose/Development/Luma3DS/sysmodules/rosalina/source/menus.c:27:
/opt/devkitpro/libctru/include/3ds/services/fs.h:476:37: note: expected 'u32 *' {aka 'long unsigned int *'} but argument is of type 'int *'
  476 | Result FSUSER_GetNandSpeedInfo(u32 *speedInfo);
      |                                ~~~~~^~~~~~~~~
/home/midnoclose/Development/Luma3DS/sysmodules/rosalina/source/menus.c:120:46: error: request for member 'sdClkCtrl' in something not a structure or union
  120 |             u32 clkDiv = 1 << (1 + (speedInfo.sdClkCtrl & 0xFF));
      |                                              ^
/home/midnoclose/Development/Luma3DS/sysmodules/rosalina/source/menus.c:123:31: error: request for member 'highSpeedModeEnabled' in something not a structure or union
  123 |                 (int)speedInfo.highSpeedModeEnabled, SYSCLOCK_SDMMC / (1000 * clkDiv)
      |                               ^
cc1: all warnings being treated as errors
make[3]: *** [/opt/devkitpro/devkitARM/base_rules:85: menus.o] Error 1
make[2]: *** [Makefile:104: build] Error 2
make[2]: Leaving directory '/home/midnoclose/Development/Luma3DS/sysmodules/rosalina'
make[1]: *** [Makefile:17: rosalina] Error 2
make[1]: Leaving directory '/home/midnoclose/Development/Luma3DS/sysmodules'
make: *** [Makefile:34: sysmodules] Error 2
```
when I run `make` I get this output. I don't understand why this is happening, because I have all the dependencies and tools listed in the README

CrimsonMaple, 2021-07-07:
This issue is caused by building with an older build of libctru.
TuxSH added some extended functionality that isn't included in the latest stable release. (SDMMC speed reading and PTM commands)

Build and update to the latest commit and that will fix the building issues.
https://github.com/devkitPro/libctru

TuxSH, 2021-07-16:
Usually libctru and Luma3DS are coordinated, so this won't be a problem when Luma releases again

closed 2021-07-16 by TuxSH