LumaTeam/Luma3DS #1643: Luma3ds not

Slicke82, 2021-06-19:
**System model:**

[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

Slicke82, 2021-06-19:
I accidentally updated my 3ds and am now getting the failed to apply 1 firm patch error. My luma3ds is on 8.1.1 and godmode is on 1.4.2. I already tried to update all the updated files onto my SD card but it doesn’t updated the firmware. Would appreciate any help anyone can provide.

Firepal, 2021-06-20:
Also had this error after updating. I did the installation procedure for the latest version of Luma3DS, and it now works (mostly).
What instructions are you following to install Luma? Hopefully [these.](https://github.com/LumaTeam/Luma3DS/wiki/Installation-and-upgrade)

closed 2022-06-05 by suppai