LumaTeam/Luma3DS #1644: Arm 11 Core 1

kiwiiberries, 2021-06-25:
**System model:**

[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

sergioinfosec, 2021-07-07:
Hello, how are you today, buddy?

You can try to follow this guide to make sure that you're doing the update without issues

https://github.com/LumaTeam/Luma3DS/wiki/Installation-and-upgrade

But also I have something to recommend, on luma config (Power On with Select Button pressed) you can mark the option:

- Disable arm11 exceptions handlers

Check if your issues is solved after that and let us know if you need something more. 

See you around.

Oreo639, 2021-07-07:
@sergioinfosec 
You read the Luma wiki yet you seemed to have missed the part where it says `ARM11 exception handlers can be turned off, but this is highly not recommended`
It wouldn't fix this issue anyway.

Updating Luma should fix it. In the future though, please follow the issue template.

sergioinfosec, 2021-07-13:
@Oreo639 Thanks for the information. I noticed that some friends were able to solve certain problems by doing this, so I thought I might be able to help in some way.

Thanks again!

ErBlack, 2021-12-29:
#1537

closed 2022-06-05 by suppai