LumaTeam/Luma3DS #1645: Yoshi Woolly world crash

T0biasCZe, 2021-06-28:
When you enter the Rosalina menu in 3DS And then exit the menu with Btm, the system crashes
1) load the game
2) Start level
3) load amiibo (optional?)
3) open Rosalina menu
4) Exit with B

https://user-images.githubusercontent.com/44525446/123698675-b8917d00-d85e-11eb-9469-a7f113cdd504.mp4

qamichelemassa5859, 2022-02-27:
I'm tried all systems, still Playable on N2DSXL??

T0biasCZe, 2022-02-27:
tried it on latest luma on new 3ds, it still happens, but only when you take screenshot in the luma menu, not when just opening it...

qamichelemassa5859, 2022-02-27:
merino-c?

T0biasCZe, 2022-02-27:
> merino-c?

What?

qamichelemassa5859, 2022-02-28:
#1720

T0biasCZe, 2022-02-28:
Yeah it still crashes with current process "merino-c"
i also tried it without the amiibo and it also crashes

qamichelemassa5859, 2022-03-04:
> Yeah it still crashes with current process "merino-c" i also tried it without the amiibo and it also crashes

it gives an errdisp type logged while i try to save the game on a O3DS console system or a N3DS system.

T0biasCZe, 2022-03-04:
Is it physical or digital game

qamichelemassa5859, 2022-03-04:
@T0biasCZe i tried disable exception handlers you will be:
```
An error occurred (ErrDisp)

Error type:       invalid

Process ID:       1
Process name:     loader
Process title ID: 0x0000000000000000

Address:          0x0000021a
Error code:       0x00000021

Press any button to reboot.
```

qamichelemassa5859, 2022-03-25:
Closed #1645.

T0biasCZe, 2022-03-25:
1) Why, it still wasnt solved
2) you didnt close it

impeeza, 2022-03-25:
I can confirm I have:

Luma 10.3
FW 11.15.0-47U

Poochy & Yoshi's Woolly World Title ID: 00040000001A4200  Version 0 (0.0.0) Code CTR-P-AJNP

When running the game, entering on Rosalina for take a screenshot and press (B) to exit the system crash with the next error:

[image]

the screenshot is take without problem

qamichelemassa5859, 2022-03-28:
> I can confirm I have:
> 
> New 3DS [crash_dump_00000000.dmp.txt](https://github.com/LumaTeam/Luma3DS/files/8354311/crash_dump_00000000.dmp.txt)
> 
> Luma 10.3 FW 11.15.0-47U
> 
> Poochy & Yoshi's Woolly World Title ID: 00040000001A4200 Version 0 (0.0.0) Code CTR-P-AJNP
> 
> When running the game, entering on Rosalina for take a screenshot and press (B) to exit the system crash with the next error:
> 
> [image]
> 
> the screenshot is take without problem
> 
> the attached file is the dump created.
> 
> [crash_dump_00000000.zip](https://github.com/LumaTeam/Luma3DS/files/8354312/crash_dump_00000000.zip)

Now that also it has been verificated the cia file and then it appeared:
```
An error occurred (ErrDisp)

Error type:       invalid

Process ID:       1
Process name:     loader
Process title ID: 0x0000000000000000

Address:          0x00000000
Error code:       0x00000010

Press any button to reboot.
```

michelemassa5599, 2022-04-04:
> I can confirm I have:
>
> New 3DS [crash_dump_00000000.dmp.txt](https://github.com/LumaTeam/Luma3DS/files/8354311/crash_dump_00000000.dmp.txt)
>
> Luma 10.3 FW 11.15.0-47U
>
> Poochy & Yoshi's Woolly World [...]

I will tried to disable arm11 exception handlers, and now this error screen is different:
```
 An error occurred (ErrDisp)
 
 Error type:       invalid
 
 Process ID:       1
 Process name:     loader
 Process title ID: 0x0000000000000000  
 
 Address:          0x00000000
 Error code:       0x00000010
 
 
 Press any button to reboot.
```

subad31155, 2022-04-30:
> I can confirm I have:
>
> New 3DS [crash_dump_00000000.dmp.txt](https://github.com/LumaTeam/Luma3DS/files/8354311/crash_dump_00000000.dmp.txt)
>
> Luma 10.3 FW 11.15.0-47U
>
> Poochy & Yoshi's Woolly World [...]

Trying to parse the crash dump in luma exception dump parser?

suppai, 2022-06-05:
Closing as a tech support question; the issue tracker is not for tech support, try Nintendo Homebrew's assistance channels (https://discord.gg/C29hYvh).

closed 2022-06-05 by suppai

T0biasCZe, 2022-06-06:
how is this a tech support question?? this is genuine bug that happens when you do a screenshot in wooly world

2nukes, 2022-06-21:
I can confirm that I am experiencing the same, and not just with screenshots. If you open the game, open the Rosalina menu, and then press 'B', the game will crash with the error code reported above, no matter what you are doing in the menu.

I am also something very similar with Poochy & Yoshi's Woolly World. Not when in the Rosaline menu, but often when transitioning between levels. I have tried different SD cards and CIAs and done every troubleshooting process I could find with no luck. The only avenues I haven't investigated fully is it being a bug with Luma or a physical hardware problem. Seeing this issue I wanted to add my 2c to see if this is genuine issue.

[image]

'merino-c' is the process having issues.

Luma3DS 11.0 (but it was happening on 10.x as well)
Sys 11.15.0-47U

I can pull the crash logs and upload them if it would be useful.

Thanks a bunch