LumaTeam/Luma3DS #1647: Hi

zmazenta, 2021-06-29:
Hi, I already updated the luma to the latest version but I still get the same error, please help me, my console is a new 3ds xl
[image]
[image]

Shadow322, 2021-06-29:
This is not a help forum take it to gbatemp.net

sergioinfosec, 2021-07-07:
Hey, how are you today, buddy?

Your issues is very insteresting, try to do these steps:

- Power Off your 3DS
- Put the card in computer
- Make a backup from all files
- Format your card (FAT32 is required, if is +64 GB try to use Guiformat.exe)
- Put all the files back in your card
- Power On your 3DS
- Select two options on that screen:

1. Show Nand or User ...
2. Disable arm11 exceptions handlers

- Press Start to save the config.

Post here if this process solved your problem. Be safe!

Oreo639, 2021-07-07:
@sergioinfosec 
Disabling luma's exception handlers would not fix anything.

sergioinfosec, 2021-07-13:
@Oreo639 User error in the process of update or something else? I'm very curious about how to solve 3DS issues. lol

Thanks for your answer on the other post, my friend.

Oreo639, 2021-07-16:
> User error in the process of update or something else?

Idk, you'd have to ask the OP. All that I know is that they have a crash due to an "invalid instruction", which can be caused by a variety of things but the reason most regular users would run in to that is usually due to trying to launch the homebrew launcher without boot.3dsx

closed 2022-06-05 by suppai