LumaTeam/Luma3DS #1649: Crash when exiting Rosalina menu in Pokemon Rumble World development versions

mariomadproductions, 2021-07-03:
**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.14.0-46E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap (fastboot3ds)

**Luma3DS version:**

v10.2.1

**Luma3DS configuration/options:**

Default EmuNAND: (N/A)

Screen brightness: (4)

Splash: (Off)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: (N/A)

--

Autoboot EmuNAND: (N/A)

Use EmuNAND FIRM if booting with R: (N/A)

Enable loading external FIRMs and modules: (x)

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: (N/A)

--

**Explanation of the issue:**
In the Pokemon Rumble World "34356M" and "34961M" development versions, opening the menu and closing it crashes the system.

**Steps to reproduce:**

1. Open the game

2. Open the rosalina menu

3. Close the rosalina menu

**Dump file:**
[crash_dump_00000002.dmp.zip](https://github.com/LumaTeam/Luma3DS/files/6758339/crash_dump_00000002.dmp.zip)

TuxSH, 2021-07-16:
Uh, maybe this is just a game crash? A development version is what it is, a development version.

I don't think this is a Luma3ds issue and don't really want to help with leaked content

closed 2021-07-16 by TuxSH

mariomadproductions, 2021-07-17:
Good point re. it being a game crash. It should be tested on a development 3DS.

The fact that its leaked content is neither here nor there. The problem isn't related to the source of the content.