LumaTeam/Luma3DS #1650: Arm11 (core 1) undefined instruction error

EnderBossGT, 2021-07-07:
**System model:**

[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

Oreo639, 2021-07-07:
Can you fill out the issue template?

EnderBossGT, 2021-07-07:
> Can you fill out the issue template?

Sorry i got confused

Oreo639, 2021-07-07:
This crash dump is not from Dragon Quest.
Did you upload the wrong dump?

EnderBossGT, 2021-07-07:
> This crash dump is not from Dragon Quest.
> Did you upload the wrong dump? (when Luma writes the dump, it says where the dump is being written to. If you need to, you can try triggering the crash again and re-uploading the correct dump.)

I did it again as you said, now, i dont know the reason to it NOT being dragon quest stories

EnderBossGT, 2021-07-07:
Anyone pls help, i really want to play dragon quest stories. (I just found out that this problem occurs with battleminerz too)

XenobladePyra, 2021-07-07:
fyi this happens with every game on the 3ds/2ds if its the arm11 error it has something to do with the rosalina menu it happens to me as well whenever I like activate some sort of cheat or something in monster hunter generations and after a few quests i get this as well i don't really know why this happens though

XenobladePyra, 2021-07-07:
someone told me to try to disable the arm11 error with the bootup method while holding select but i can confirm right now that doesn't solve it it'll just freeze the 3ds/2ds

EnderBossGT, 2021-07-07:
> someone told me to try to disable the arm11 error with the bootup method while holding select but i can confirm right now that doesn't solve it it'll just freeze the 3ds/2ds

I know that it freezes, i tried it too.
Maybe the problem is with my CIA file?

EnderBossGT, 2021-07-07:
> This crash dump is not from Dragon Quest.
> Did you upload the wrong dump? (when Luma writes the dump, it says where the dump is being written to. If you need to, you can try triggering the crash again and re-uploading the correct dump.)

Hey uhh i just remembered that the name isn't dragon quest stories but its monster hunter stories 🤦🤦

EnderBossGT, 2021-07-07:
> someone told me to try to disable the arm11 error with the bootup method while holding select but i can confirm right now that doesn't solve it it'll just freeze the 3ds/2ds

Hey uhh i just remembered that the name isn't dragon quest stories but its monster hunter stories 🤦🤦

closed 2022-06-05 by suppai