LumaTeam/Luma3DS #1651: N3DS Crash while playing SM64

Rocketter, 2021-07-07:
NEW 3DS

**SysNAND version (+emu/redNAND version if applicable):** 11.14.0-46E

**Entrypoint (How/what you're using to boot Luma3DS):** boot9strap

**Luma3DS version:** 10.2.1

**Luma3DS configuration/options:**

Default EmuNAND: ( ) none

Screen brightness: ( ) 3

Splash: ( ) default

Splash duration: ( ) default 

PIN lock: ( ) none

New 3DS CPU: ( ) default
--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( ) none

Enable loading external FIRMs and modules: ( ) default

Enable game patching: ( ) default

Show NAND or user string in System Settings: ( ) default

Show GBA boot screen in patched AGB_FIRM: ( ) default 

Set developer UNITINFO: ( ) default 

Disable Arm11 exception handlers: ( ) default

Enable Rosalina on SAFE_FIRM: ( ) default
--

**Explanation of the issue:**

While playing SM64 on N3DS crashes after a certain amount of time.

**Steps to reproduce:**

1.

2.

**Dump file:**

[crash_dump_00000013.zip](https://github.com/LumaTeam/Luma3DS/files/6779184/crash_dump_00000013.zip)

Herbin12, 2021-10-04:
Are you playing on an emulator, CIA app, what is it? We need more information to help.

Rocketter, 2021-10-04:
> Are you playing on an emulator, CIA app, what is it? We need more information to help.

Its an N64 Inject (so .CIA I think) on the New 3DS, I dont know the specific emulator if there is one.
I dont have my n3ds with me right now Inwill give you more info when I Can.

urherenow, 2021-10-04:
SM64 is the N64 game reverse-engineered, then compiled to run natively, so it is sort of home brew, but not emulation. I've compiled it myself for the PC, Switch, and even to run in a browser. You must run a batchfile on a good dump of the N64 game, to extract the assets and such, in order to compile it.

closed 2022-06-05 by suppai