LumaTeam/Luma3DS #1654: Black screen on Pkmn Omega Ruby

dynux90, 2021-07-11:
Model: New 2DS XL
fastboot3ds v1.2
Luma3DS v10.2.1

Content from: errdisp.txt

Error type:       generic

Process ID:       39
Process name:     sango-1
Process title ID: 0x000400000011c400

Address:          0x0012884c
Error code:       0xd900458b

fsilletti, 2021-07-12:
same problem

TuxSH, 2021-07-16:
Are you using any kind of patches?

dynux90, 2021-07-17:
> Are you using any kind of patches?

Nope, also tested on a new sdcard

notsunburn, 2021-07-17:
mine has been doing the same thing but its not just in the game its when I turn my 3ds on. But it also does do it in game, like 3 secs after trying to use a a cheat from checkpoint. please get back to me with some help on the problem. Thanks

dynux90, 2021-07-17:
I do not use cheats, i also tried with cheats to walk over wall to avoid the ecounter that makes me crash the game but the cheats also do not work

notsunburn, 2021-07-17:
and the only way it doesn't show me "An exception occurred" is when i dont have the sd card in the system

closed 2022-06-05 by suppai