LumaTeam/Luma3DS #1655: Random blacks screens / unstable sleep mode

AntoineChabanel, 2021-07-13:
**System model:**

[Old 2DS]

**SysNAND version (+emu/redNAND version if applicable):**

[11.14.0-46E]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap]

**Luma3DS version:**

[10.2.1]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (After Payloads)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ()

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

Hello !

There is some bugs that have appeared when i installed b9s and luma3ds on my old 2ds.

-Sometimes I need to spam the power button to start my 2ds.
-When it's on, my 2ds randomly shows a black screen (during less than 1 second) then continues to work. It happens several times. It usually happens, and most of the time happens when i enter & exit sleep mode, but can also happen completely randomly.
It never crashes tho.

This is really weird, I tried to found a similar problem on the internet but found nothing.

I (think) have followed every step correctly when I installed luma (it was a few months ago).

Thanks, have a good day/night !

wasago01, 2021-12-08:
I have same problem Instaled custom firmware today and geting this random black screen.
Same old 2ds

yzabeast1, 2022-04-06:
I modded my 2ds a few months ago with gm9, luma3ds, and boot9strap and just recently started having this issue

AntoineChabanel, 2022-04-06:
Apparently it was just an issue with the sleep button. I recommand switching it on/off 10-20 times in a row to see if it still happens (maybe try to clean it). @yzabeast1 @wasago01

closed 2022-04-06 by AntoineChabanel