LumaTeam/Luma3DS #1656: Unsure what to do with this

Azzythecomic03, 2021-07-15:
Ok so hear me out. I'm not really up to date with all of this and I love to learn more as to how to use my modded 3DS but I don't know what to do with this and I think this happened because I don't have luma3ds on my New Nintendo 3DSXL (Red coat original) and I would love to receive steps as to how I can fix this issue I'm going to send a photo also I use Mac for all of my extracts if I have to extract anything so if there's a way to do this with steps that require a Mac that would be entirely awesome and pardon me for my insignificance with knowledge of this class cause all of this is truly new to me and I would love to learn more. Here I post a picture of the problem in which I don't understand.
[image]

Oreo639, 2021-07-15:
There isn't much anyone can really help you with without more information as for when that crash occurs and what you did before it started happening. (also, in the future, please follow the issue template)

Azzythecomic03, 2021-07-15:
The crash occurs when you boot up Nintendo music. This originally leads to
fbi and luma3ds and stuff like that. I got this version of the mod done by
a guy in Puerto Rico and although he explained it very well to me I didn't
understand him because I was still new to this while thing. And before this
was even an issue I would normally set up my games and download with via
fbi and even trying out the gba emulation that the guy installed too.
Surprising that all of that was just all in one app from Nintendo I found
that to be pretty cool until of course after a few years ago since I got it
modded that happened and since last year I didn't know what it could be or
what it was. (Sorry for not using the template I'm new even for GitHub in
future issues I will learn the template to give out the issue better pardon
me for not using it)

Oreo639, 2021-07-15:
> This originally leads to fbi and luma3ds and stuff like that.

Just to clarify, Luma3DS and Luma Updater are not the same thing.

Two things:
1. What is the version number displayed when booting while holding select?
2. What happens when you try booting the 3ds without the SD card? (Does it boot? If so, what happens when you try to launch 3DS Sound?)

Medden63, 2021-08-06:
I have the same problem, when starting the console after a short time, a crash screen appears.

Oreo639, 2021-08-06:
*a crash screen* can mean a lot of different things depending on the information displayed on that screen.

closed 2022-06-05 by suppai