LumaTeam/Luma3DS #1658: Arm11 Crash

XiNaru, 2021-07-28:
I completely reset my 3ds due to the following problem and redoed the hack. All the latest version.

but when I try to start a game (in this case Monster Hunter Stories) I get an Arm11 crash.

I use the New 3DS XL with firmware 11.15.0-47E and Luma3DS 10.2.1

Oreo639, 2021-07-28:
How did you install Monster Hunter Stories? If it is from the eshop, did you try doing a repair install?

XiNaru, 2021-07-28:
no it was an BACKUP installation through FBI

Oreo639, 2021-07-28:
Then redump it, otherwise, piracy isn't supported.

XiNaru, 2021-07-28:
not piracy, only BACKUP
that was an backup dump from my modul

SpacyJacobson, 2021-08-03:
I have the same error but I downloaded it from the Internet

[crash_dump_00000000.zip](https://github.com/LumaTeam/Luma3DS/files/6923403/crash_dump_00000000.zip)

closed 2022-06-05 by suppai