LumaTeam/Luma3DS #1662: Error Loading Games

IgnacioLeonI, 2021-08-11:
**System model:**
New 2DS XL

**SysNAND version (+emu/redNAND version if applicable):**
11.15

**Entrypoint (How/what you're using to boot Luma3DS):**
B9S

**Luma3DS version:**
10.2.1

**Luma3DS configuration/options:**

Screen brightness: (4)

Splash: (Off)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X )

Show NAND or user string in System Settings: ( X)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

After the game start, (after the initial title screen) it gives me this error:

In M&L BIS: Error Remillio (after the initial cinematic)
In LM : Error greencube (after you choose the game file)

Sorry if this is an error that should not be solved here. I would be grateful if someone could tell me where to look to solve this. It was my gift for my kid for the children day :(.

[image]
[image]

Oreo639, 2021-08-11:
How did you install these games? The errors indicate that either the game files or the save is corrupted.

IgnacioLeonI, 2021-08-11:
> How did you install/run these games? The error codes indicate that either the game files or the save is corrupted.

Directly from the main menu of the console. I also did the "import seed" with the FBI

Oreo639, 2021-08-11:
Is this installed from the eshop or is this being run from a cartrige is what I meant?

IgnacioLeonI, 2021-08-11:
> Is this installed from the eshop or is this being run from a cartrige is what I meant?

Downloaded from a webpage and installed through cia

Shadow322, 2021-08-11:
> > Is this installed from the eshop or is this being run from a cartrige is what I meant?
> 
> Downloaded from a webpage and installed through cia

Luma3ds does not support or Condon piracy

Doxerythm, 2021-08-16:
> > Is this installed from the eshop or is this being run from a cartrige is what I meant?
> 
> Downloaded from a webpage and installed through cia

Ok, i'm trying to report you to the authorities.

IgnacioLeonI, 2021-08-16:
:(

closed 2021-08-16 by IgnacioLeonI