LumaTeam/Luma3DS #1664: Game patching doesn't appear to do anything

AlgebraManiacABC, 2021-08-17:
**System model:** Old 3DS XL

**SysNAND version:** 11.14.0-46U    SysNAND

**Entrypoint:** Boot9Strap

**Luma3DS version:** v10.2.1

**Luma3DS configuration/options:**
Screen brightness: (4)
Splash: (Off)
Splash duration: (1)
PIN lock: (Off)
Enable loading external FIRMs and modules: (  )
Enable game patching: ( x )
Show NAND or user string in System Settings: ( x )
Show GBA boot screen in patched AGB_FIRM: (  )
Set developer UNITINFO: (  )
Disable Arm11 exception handlers: (  )

---

**Explanation of the issue:**

I have been attempting to modify Ace Attorneys Trilogy's music by making all music files the same thing (to find out what files are what). I have put these music files (all the same file, but labeled differently as to appear normal) in the romfs/sounds/stream folder and double checking the title ID is 00138f00. I have triple checked nearly everything but the game still appears unmodified.

**Steps to reproduce:**

1. Place modified music files in romfs

2. Boot up game with listed Luma settings

Oreo639, 2021-08-21:
That is the titleid low. You need the full titleid including the `00040000`.
For example:
`sd:/luma/titles/0004000000138F00/romfs/sounds/stream`

AlgebraManiacABC, 2021-08-22:
Oh thanks! The patching does do something now... But it gives a crash dump x.x

[crash_dump_00000002.txt](https://github.com/LumaTeam/Luma3DS/files/7028205/crash_dump_00000002.txt)

(Couldn't upload the .dmp directly, so I just changed the extension, you'll have to change it back)

Should I open a new issue for this dump?

AlgebraManiacABC, 2021-08-23:
Update / More info:
I wasn't sure if I was supposed to include ALL files from romfs or ONLY modified files, so I tried with both (both crashed).
I then decided to simply put the unmodified romfs folder into the path, which also crashed...

Shadow322, 2021-08-23:
This isn't a help forum use gbatemp.net for this kind of thing

AlgebraManiacABC, 2021-08-23:
Apologies...

closed 2021-08-23 by AlgebraManiacABC