LumaTeam/Luma3DS #1665: Suggestion: getting supported cheats feature more aligned with CTPRF plugin

monkeyman4412, 2021-08-20:
CTPRF supports a few features that the current cheat system doesn't. Primarily storage registers and I think the data registers as well. This causes some headaches as not all codes that were written in CTPRF work with the cheat system that luma has.

monkeyman4412, 2021-08-20:
https://gist.github.com/Nanquitas/d6c920a59c757cf7917c2bffa76de860 
going to link the github so it's clear. What I'm talking about is
"
You have access to:
 - 2 offset registers (not persistent)
 - 2 data registers (not persistent)
 - 2 storage registers (persistent)"
 The storage registers I am certain is not supported, while the data register I'm not sure, I'll have to recheck.

KevinCathcart, 2021-08-22:
I’m a little confused by this. The code for the cheats menu most definitely includes support for the persistent storage registers, as well as the two data registers. As far as I know, the only missing feature is the custom asm. (Which is not trivial to implement). 

They have been implemented since v10.0.

Now, it is entirely possible that some of these features are not working correctly. If that is the case it can and should be fixed.

monkeyman4412, 2021-08-23:
said features may just be broken. Going to write a few test codes and see specifically what doesn't work or isn't working correctly. Since already I used a code (that's known to be working in CTRPF) and that just, didn't work in luma.
 I'm using three features that could be going wrong. The storage register, data registers, or conditional mode. As I tend use conditional mode to compare between storage register usually and offset.

monkeyman4412, 2021-08-23:
is data register operations broken? I have a simple code (this is bowser inside story 1.2. USA)
It may not work the same way since it's not looking for a pointer.  I'll have some video to show the issue in a bit. All it does is tell the data register add 1 (which is going to always start from 0)
On ctrpf it works. making syrup display 0. While on Luma's cheat system, nothing happens.
[test]
D3000000 30000000
DD000000 00000040
D4000000 00000001
D6FB1C44 00FB1C44
D2000000 00000000

monkeyman4412, 2021-08-23:
okay after a bit of testing. It's conditional mode that's not working in Luma. I thought originally it was the data registers or storage, however I was clearly wrong.

monkeyman4412, 2021-08-23:
[conditional example]
D3000000 30000000
DFFFFFFF 00000003
80000000 00000020
DB000000 00FB1BF0
D4000000 00000001
D7000000 00FB1BF0
D5000000 00000000
DF000002 00020000
D0000000 00000000
DF000002 00010000
D4000000 00000001
DF000002 00020000
So this is for bowser's inside story USA 1.2. the item it can be changing is likely going to be a little random since no pointer. But on my end here. it slowly updates an item. In luma, it does nothing.

KevinCathcart, 2021-08-23:
I think pull request #1623 is intended to fix the conditional operations.

closed 2022-03-09 by TuxSH