LumaTeam/Luma3DS #1666: Arm 11 Crash Pokémon XY and ORAS

servalsanz, 2021-08-23:
Hello, I have been desperate for a long time, I have been with this error since I purchased my new 2ds console.

The console already had CFW installed.

When I had it I updated everything to the latest version. All games seem to be working fine, but when playing Pokémon games, I get an ARM11 error (code 0) and each time I get a different error, as seen in the photos.

I have reinstalled, updated, changed the files ... the same thing always happens to me, an error in a random place (it does not usually occur in the same place) but constant (whatever I do happens)

The weird thing is that it not only happens to me with the backup, it also happens to me with the original game cart.

Does anyone know what the error may be and how to fix it?

[image]
[image]
[image]

RealKGB, 2021-08-28:
This works fine for me with X and Y carts.
Try reinstalling the CFW? Perhaps it got corrupted.

N-rotceh, 2021-09-02:
I have everything updated to the latest version and I have the same issue with Pokemon A. Sapphire and Pokemon X as well, I'm using the updated version of both of them.

Squall-Leonhart, 2021-09-03:
follow the guides to update/migrate from previous cfw to current boot9strap 

https://3ds.hacks.guide/

closed 2022-06-05 by suppai