LumaTeam/Luma3DS #1671: Idea: Allow ZL, ZR, C-stick to be used to open the Rosalina menu

RealKGB, 2021-08-28:
As users of Luma3DS are hopefully aware, you can open the Rosalina Menu at any time by pressing a button combo. It's (by default) set to L+Down+Select, but this can be changed. However, on a New-series 3DS console, the ZL, ZR, and C-stick cannot be used for the combo.

Since almost no games use the ZL/ZR buttons or the C-stick (I think Smash 3DS uses the C-stick but not sure), I was hoping to set the Rosalina Menu combo to ZL+ZR, that way I don't accidentally open it while playing a game.

Would it be possible to add the functionality for the Rosalina Menu to allow the ZL/ZR buttons and the C-stick for the "open-the-Rosalina-menu" combo?

I'm fairly certain that Luma3DS can tell between a regular 3DS and a New 3DS since it has the "New 3DS settings" menu, so could this be implemented?

Nuck-TH, 2021-09-08:
It is difficult(if possible at all from arm9), because despite being integrated into system, C-stick and ZL/ZR on New3DS are still separate thing like Circle pad pro was and are accessed same way.

RealKGB, 2021-09-08:
Interesting. I'd have thought that they'd integrate it like the other controls instead of treating like another module and tacking it on.
Thanks for the info!

closed 2021-09-08 by RealKGB