LumaTeam/Luma3DS #1673: For people with broken parts using Rosalina Menu

RemittentBow410, 2021-09-02:
Can you make some type of configurable file so people can choose the configuration to use on Rosalina menu so it is accessable. I have a broken Select button and as it stands I cannot find any way to access Rosalina menu without fixing the broken part (which I understandably don't want to do if at all possible) I would really appreciate it :D

Weigurde, 2021-11-10:
somebody close this?

wuniversales, 2022-02-01:
> Can you make some type of configurable file so people can choose the configuration to use on Rosalina menu so it is accessable. I have a broken Select button and as it stands I cannot find any way to access Rosalina menu without fixing the broken part (which I understandably don't want to do if at all possible) I would really appreciate it :D
> 
> (edit) there is already a config.bin file that you can do this, sorry lol

Please close the ticket if it has been fixed.
Greetings

TuxSH, 2022-03-08:
https://3ds.eiphax.tech/config

closed 2022-03-08 by TuxSH