LumaTeam/Luma3DS #1674: exception occured arm11 core 0 data abort permission page

gulp1337, 2021-09-05:
**System model:** New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**
11.15.0-47E

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap

**Luma3DS version:**
v10.2.1-58540cb2

**Luma3DS configuration/options:**

Screen brightness: (4 )

Splash: (Off) 

Splash duration: (1) 

PIN lock: (Off) 

New 3DS CPU: (Off) 

Enable loading external FIRMs and modules: ( )

Enable game patching: (X )

Show NAND or user string in System Settings: (X )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

**Explanation of the issue:**
Everytime i play Minicraft and try to save it this pops up:
An exception occured 
Processor: Arm11 ( core 0 )
Exception type: data abort
Fault status: Permission - Page
Current process: 3dsx_app (000400000D21E00)

**Steps to reproduce:**
Open Homebrew Launcher -> launch Minicraft 3DS homebrew edition -> save

**Dump file:**

[crash_dump_00000004.zip](https://github.com/LumaTeam/Luma3DS/files/7111350/crash_dump_00000004.zip)

Squall-Leonhart, 2021-09-06:
you might have damaged nand on the mainboard or a faulty cfw install.

gulp1337, 2021-09-07:
> you might have damaged nand on the mainboard or a faulty cfw install.

alr thanks for responding

gulp1337, 2021-09-07:
> you might have damaged nand on the mainboard or a faulty cfw install.

i don't think thats the case cuz everything else is running totally fine, minicraft is the only thing not running as intended

Herbin12, 2021-10-04:
that sounds like a software issue then, not a luma3ds issue.

closed 2021-10-05 by gulp1337

Dumbelfo08, 2023-07-31:
This also happens to me, but only about 70 % percents of the time