LumaTeam/Luma3DS #1676: just more Arm 11 crashes

TroyTT4, 2021-09-12:
I've been getting arm 11 crashes on my New3Ds for a while, and I thought it was just me, but after checking this page I can see that everyone has been getting arm 11 crashes. please fix these issues soon

Oreo639, 2021-09-12:
You seeing the exception handler when an application crash occurs is not a bug.
Crashes can happen for a wide variety of reasons, most of them aren't issues with Luma.
There is a link in the issue template to places where you can ask for help, however if you are certian that this crash is caused by an issue in luma or rosilina, you can try reporting that issue.

TroyTT4, 2021-09-12:
also sometimes my 3ds will go into a state where it will give a crash when I try and turn in on so I can't use it until it goes back to normal.

the crashes are;
Processor: arm 11 (core 1)
Exception type: prefetch abort
Fault status: debug event
Current process: fs (0000000000000000)

then it said

Processor: arm 11 (core 1)
Exception type: undefined instruction
Current process: sm (0000000000000000)

then it said

 "unable to mount CTRNAND or load the CTRNAND FIRM
  Please use an external one."

and now it wont even turn on. the blue light for the power button turn off right after turning on. if I try turning it on with the SD card out of it I get the the third "an exception occurred" message.

Squall-Leonhart, 2021-09-17:
your 3DS has a hardware fault.

TroyTT4, 2021-09-26:
> your 3DS has a hardware fault.

how do I fix it (if that's even possible)

TroyTT4, 2021-10-08:
@TheCatDev what exactly does the ctrtransfer do? the instructions for it and the finalizing setup just look like how luma 3ds and the features with it are installed.

TheCatDev, 2021-10-09:
Ctrtransfer can unbrick consoles.
Its like flashing another NAND.

TroyTT4, 2021-10-09:
then I probably should've mentioned a lot earlier that even after not turning on, it did again after a few days, so it's not permanently bricked. I wish I could actually fix the constant crashing and temporary bricking.

TheCatDev, 2021-10-09:
Try reformating the sd card or try another sd card.

TroyTT4, 2021-10-09:
I tried another SD card and it didn't change anything. what's reformatting one?

closed 2022-03-08 by TuxSH